parallella/parallella-hw #10: Incorrect directory tree for Verilog files in parallella_7020_headless.tcl

closehaulcom, 2015-03-29:
SHA: a826fbb71392e5ea311d6619f166b6af69e7c3dd

In parallella-hw/fpga/vivado_projects/parallella_7020_headless/parallella_7020_headless.tcl,
lines 97-127 point to a non-existent directory $origin_dir/../../hdl

It looks like some of the original directory tree still exists in parallella/hw-fpga/old and most of the Verilog source files have been moved to the ../../src directory, but the files in the  ./parallella-I sub-directory appear to be missing in the new file structure.

lstewart, 2015-04-07:
Also, the fpga/old/hdl/parallella-I/parallella.v doesn't seem right, as it gets syntax errors due to duplicate
input declarations for multiple emaxi_ signals

hewittc, 2015-04-07:
It seems like an incomplete commit. There is also evidence of a failed merge at src/gpio/hdl/parallella_gpio_emio.v:31

<<<<<<< HEAD
   GPIO_I,
   // Inouts
   GPIO_P, GPIO_N,
   // Inputs
   GPIO_O, GPIO_T
   );

   inout [`GPIO_NUM-1:0] GPIO_P;
   inout [`GPIO_NUM-1:0] GPIO_N;

   output [63:0]  GPIO_I;
   input  [63:0]  GPIO_O;
   input  [63:0]  GPIO_T;

   wire [`GPIO_SIGS-1:0] gpio_i;
   assign GPIO_I[`GPIO_SIGS-1:0] = gpio_i;
   wire [`GPIO_SIGS-1:0] gpio_o
                         = GPIO_O[`GPIO_SIGS-1:0];
   wire [`GPIO_SIGS-1:0] gpio_t
                         = GPIO_T[`GPIO_SIGS-1:0];
=======
   PS_GPIO_I,
   // Inouts
   GPIO_P, GPIO_N,
   // Inputs
   PS_GPIO_O, PS_GPIO_T
   );

   parameter  NUM_GPIO_PAIRS = 24;       // 12 or 24
   parameter  DIFF_GPIO      = 0;        // 0 or 1
   parameter  NUM_PS_SIGS    = 64;
>>>>>>> origin/elink_redesign_fred

aolofsson, 2015-04-07:
Yes...this is still work in progress. The archive is really the snapshot for now. I am working on getting the verification done to put this in complete source form. One of the things that had to be done was to move away from the Vivado block based designer to a pure verilog design. This work has been done. The entry point for all of this is "fpga/src/elink/hdl/elink.v". There will be a branch and tag once done.. (and an update project file).

lstewart, 2015-04-07:
For what I’m trying to do, the pure verilog is best.  I am trying to merge this
design with other components in Bluespec, which is all batch rather than interactive.
-Larry

aolofsson, 2015-04-09:
Closing this issue. More changes going on, things will stabilize next week. If by next Friday things have not settled, then let me know.

closed 2015-04-09 by aolofsson