parallella/parallella-hw #11: Split out RTL files to separate repo

olofk, 2015-03-30:
Hi,

I was revisiting adding FuseSoC support for parallella and all its support cores and noticed the refactoring of the repo. It would be great to split out the RTL files to a separate (epiphany-rtl? parallella-rtl?) repo for several reasons. It would make it easier for non-parallella boards to reuse the Epiphany-facing RTL and it will save users from pulling in all the Kicad files, old prototype board projects and other things.

Now would probably be a good time as most people probably still haven't updated after the refactoring. If you want a unified repo this could still be managed by importing the RTL repo as a git submodule

aolofsson, 2015-03-30:
Good suggestion, could you elaborate on the ideal directory structure for fusesoc? Is this it?
https://github.com/openrisc/orpsoc-cores
btw. why split parallella rtl and epiphany rtl? Wouldn't it make more sense to have two repos as: 1.) RTL, 2.) HW?  This might be reasonable....

The reason for the current structure was having too many repos and syncing between them. Taje the ESDK build process is an example of something that is quite complicated... (tons of repos to tag and keep track of to make sure things move in lock step).

Comments?

olofk, 2015-03-31:
I didn't mean to split up epiphany and parallella code. This was more intended as different name suggestions. There is however some reasons for splitting it up further (I have spent quite a lot of time on Open Source IP managment over the last years). The biggest one would perhaps be the ability to make new tagged releases on separate components, but I'm also aware of the added complexity that comes with a lot of repos. so for Parallella, a single RTL repo is probably the best way.

The exact structure is not that important for me, as one of the design rules for FuseSoC is to not force any particular structure on other projects RTL code. That said, I think the current structure looks good with all the cores such as emmu, memories etc residing in sibling directories. The top-level projects (the vivado_projects directory) could be dropped in there side-by-side with the other stuff.

aolofsson, 2015-04-09:
Olof,
I have thought about it and I think we will keep it the way it is....old obsolete stuff will be removed from the directory to make it simpler. Compared to anything related to GNU or linux, pulling in a couple of board files should not be a a big deal hopefully. Git clone takes a few seconds longer. To me it's more important to have a small set of central repos than to manage 20 different ones (which we have already for Parallella and Epiphany). We should be able to import everything as a submodule.  All source code is under "fpga/src". I will look into that.

aolofsson, 2015-04-22:
I have split out the pure source HDL into a separate repo. I agree with you, it's the right thing to do.
https://github.com/parallella/oh

closed 2015-04-22 by aolofsson