parallella/parallella-hw #18: A clean "no-ip" version of elink

aolofsson, 2015-04-09:
Create a clean elink design that is completely free of binary IP. We did this with the first design of the elink 3 years ago, but due to performance reasons we had to moved towards using FPGA IP. It would be fantastic to have a truly portable clean implementation of the elink that is not reliant on the vendor specific IP.

Suggestion is to use `define TARGET_OPEN as the keyword. The blocks that need to be implemented include:

single ported memory
dual ported memory
sync fifo
async fifo
serdes for rx and tx

olofk, 2015-04-09:
Perhaps it would be a good start to split out the elink stuff to a new repo. Once it's in good shape it can always be fed back to the parallella-hw repo if you want it there

sbourdeauducq, 2015-04-10:
Migen has portable abstractions for all of those things except the last. Just saying...

cfelton, 2015-04-14:
@cfelton watching

aolofsson, 2015-04-14:
sbourdeauducq,
Can you elaborate. Not familiar with migen. Does it generate Verilog models? Is this the one? https://github.com/m-labs/migen

cfelton, 2015-04-14:
@aolofsson, you are going to get all kinds of solicitation :) 

 [MyHDL](https://github.com/jandecaluwe/myhdl) would be a much better option, here is an [example replacing the USRP GEMAC IP specific FIFOs with agnostic versions](https://github.com/cfelton/test_gemac/tree/master/alt_fifo).  You can find replacements for most of the IP, albeit probably not all in one spot.

aolofsson, 2015-04-14:
Chris,
Thanks! MyHDL is great, but my impressions is that the gain is really for more complex models.  Tell me if I am wrong?

What I really need at this point are simple verilog models. Given how simple these are, not sure MyHDL will help here? You still have to read the Xilinx datasheets. I was hoping that someone had already done this before?

BUFG.v 	Adding stubs files for xilinx IP 	4 months ago
	BUFIO.v 	Vivado run through 	5 days ago
	BUFR.v 	Adding stubs files for xilinx IP 	4 months ago
	IBUFDS.v 	Adding stubs files for xilinx IP 	4 months ago
	IBUFGDS.v 	Adding stubs files for xilinx IP 	4 months ago
	ISERDESE2.v 	Major cleanup, refactoring, and feature completion 	3 days ago
	OBUFDS.v 	Adding stubs files for xilinx IP 	4 months ago
	OBUFTDS.v 	Adding stubs files for xilinx IP 	4 months ago
	ODDR.v 	Vivado run through 	5 days ago
	OSERDESE2.v 	Vivado run through 	5 days ago
	PLLE2_BASE.v 	Major cleanup, refactoring, and feature completion 	3 days ago
	RAM32X1D.v 	Adding stubs for IP blocks used in elink 	5 days ago
	axi_bram_ctrl_16b.v 	Adding stubs for IP blocks used in elink 	5 days ago
	fifo_async_103x 	Adding stubs for IP blocks used in elink 	5 days ago
	fifo_async_103x16.v 	Adding stubs for IP blocks used in elink 	5 days ago
	fifo_async_103x32.v 	Adding stubs for IP blocks used in elink 	5 days ago
	memory_dp_48x4096.v 	Adding stubs for IP blocks used in elink 	5 days ago

cfelton, 2015-04-14:
@aolofsson, correct MyHDL doesn't directly help, someone would have needed to create the models.  The FIFOs can be replaced with agnostic versions (BRAM inferred), similar to the above example link.  The other primitives, if not reliably inferred, models can be created to instantiate the correct primitive depending on the target.  MyHDL can be used to manage this complexity (e.g. selecting primitive for different technologies).  If it sounds of interest I can provide some examples.

sbourdeauducq, 2015-04-14:
Yes, it's m-labs/migen. You can get verilog output from it indeed. Here is a complex example of using asynchronous FIFOs (which are part of the Migen libraries):
https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/core.py
and simpler ones with synchronous FIFOs:
https://github.com/m-labs/migen/blob/master/migen/test/test_fifo.py
Migen docs are here: http://m-labs.hk/gateware.html

sbourdeauducq, 2015-04-14:
Oh and also OpenVizsla are using Migen and made a few nice writeups about it:
http://debugmo.de/2014/09/ov3-fpga-helloworld/
http://debugmo.de/2014/08/ov3-fpga-design/

aolofsson, 2015-04-17:
Just committed a working "no-ip" version of the elink. Needs more validation, but it's a start. (wrote the functions by hand).

closed 2015-04-17 by aolofsson