parallella/parallella-hw #27: Porcupine PCB

Nobis99, 2015-04-17:
I am attempted to get a PCB printed. However, I am having difficulty with the current Kicad project file and gerber files. I am unable to produce the files in a format that I can submit for printing. 

Would it be possible to provide the files in a format that they could be accepted by a PCB manufacturer, like OSH Park?

Very Respectfully.

aolofsson, 2015-04-18:
The porcupine-cad.zip has all the gerbers needed for creating a PCB. What is the problem?

Nobis99, 2015-04-20:
Mr. Olofsson,

Thank you very much for your fast response. I attempted to open the gbr
files with gerbv and gEDA, but was unable.

Further, I attempted to submit the files to OSH Park for printing. However,
errors were generated. Attached is a screenshot generated by the program.
My apologies, if I am not submitting in the proper format. Any help would
be greatly appreciated.

I have been on the waiting list from Digikey for over one-month. I would
very much like a porcupine board to finish my product evaluation for my
employer.

Kind regards,
Kane

[image: Inline image 1]

Nobis99, 2015-04-26:
Mr. Olofsson,

If there is a preferred method for me to submit the files to a printer, I will happily use another vendor to purchase a printed PCB. Do you have a suggestion?

Kind regards,
Kane

aolofsson, 2016-02-04:
Closed. Files provided as is

closed 2016-02-04 by aolofsson