parallella/parallella-hw #30: the IP address of P1601

yyddddmw, 2017-06-15:
Hello every:
        I bought a P1601 and according the "Quick Start Guide" to do it. But I do not know the ip of P1601, so I can not login it.

olajep, 2017-06-15:
Hi yydddmw,

From the release notes:
https://github.com/parallella/parabuntu/releases/tag/parabuntu-2016.11.1

First boot

Hostname: "parallella", "parallella." (notice trailing period) or
"parallella.local", depending on your operating system.
IP address: The IP address is acquired from DHCP. 10.11.12.13 will be used as
fallback.
Username: parallella
Password: parallella

If that doesn't work you should be able to find the board IP in the DHCP client
list in your router's web ui.

// Ola

yyddddmw, 2017-06-16:
Hi olajep,

Thank you for your reply.
According the release notes, the connection and the login is successful.
          
Thanks.